Carrot-collective/Blobifier #5: Servo_arm_???.stl

evelyndooley, 2024-01-29:
Wondering where these files are. They're mentioned in the README but don't seem to appear in the repo. May have forgotten a `git add` ;)

Dendrowen, 2024-01-29:
Oh, your absolutely right! I'll add them. I didn't think of it anymore because I'm using the provided servo arm.

closed 2024-01-29 by Dendrowen

ksteddom, 2024-09-11:
I'm having trouble getting the Servo_Arm.stl to fit a MG90S servo. It sticks up too high and interferes with the tray sliding. I have verified that the servo is completely seated in the slot and the arm is pressed all the way down. @Dendrowen do you mean to say you are using one of the servo arms that comes with the servo or the newer Servo_Arm.stl file?

Dendrowen, 2024-09-12:
I'm using the included servo arm atm. That one seems to work fine for me. Did you print it in the provided orientation?

ksteddom, 2024-09-12:
Thanks for the clarification. I thought I printed it correctly. I’m going to print it again to double check. If I have any more issues  I’ll start a new thread.