Carrot-collective/Blobifier #7: `Blobifier.cfg` has incorrect indenting

ejsears, 2024-02-19:
The following lines are incorrectly indented in the `blobifier.cfg` file and result in the following errors:

```
Source contains parsing errors: '/home/eric/printer_data/config/hardware/blobifier.cfg'
	[line 356]: 'M220 S{backup_feedrate * 100}\n```
```

Line 356: `M220 S{backup_feedrate * 100}`

```
Option '{% set raise_z_min' is not valid in section 'delayed_gcode blobifier_init'
```

Line 467: `{% set raise_z_min = (bl.iteration_z_raise - ((bl.max_iterations_per_blob - 1) * (bl.max_iterations_per_blob) / 2) * bl.iteration_z_change) %}`

```
Option 'blobifier_servo pos' is not valid in section 'delayed_gcode blobifier_init'
```
Line 486: `BLOBIFIER_SERVO POS=out`

ejsears, 2024-02-19:
Opened a PR at https://github.com/Dendrowen/Blobifier/pull/8 if it is easier for you.

Dendrowen, 2024-02-19:
Merged. Ty

closed 2024-02-19 by Dendrowen