Carrot-collective/Blobifier #10: kinematic mounts

jckennell, 2024-03-01:
for the bases for the people with kinematic bed mounts, there are 2 poplular sets that people use. the mounts from Mandela Roseworks sits the bed higher than the whoppingpochard kinematic mounts. i modified mine to 55.4mm tall (16.7mm taller than the original base). also the shaker arm needs to be 16.7mm taller for beds that sit higher as well.
[image]

Dendrowen, 2024-03-05:
Thanks! Did you create this in the .step or .f3z files?

jckennell, 2024-03-05:
I used your .step file. My Fusion skills aren't very good, but basically just added a block on top of the arm and and put the holes in it. I should have saved it, but I didn't because I'm using Fusions free hobby license, which only let's me save so many files. So the ones that get deleted are the most simple ones to recreate if I have to. I just posted the Pic for reference as to what I did for that part.Sent from my T-Mobile 5G Device

Dendrowen, 2024-03-07:
Makes sense. Do you still have the .STL, and if so. could you create a pull request or send it to me on Discord?

jckennell, 2024-03-07:
I messaged you on the discord. Github is witchcraft to me. I've tried adding files, pull requests, branches, forks. It's all beyond me. LolSent from my T-Mobile 5G Device

Dendrowen, 2024-03-07:
Files are added. Thank you very much!

closed 2024-03-07 by Dendrowen