Carrot-collective/Blobifier #11: .step does not include tray part

53Aries, 2024-03-09:
Just fyi. Thanks.

Garr-Garr, 2024-05-01:
It looks like it's included:

> https://github.com/Dendrowen/Blobifier/blob/main/Source%20files/Blobifier.step

[image]

53Aries, 2024-07-02:
> It looks like it's included: [...]

I don't see it. By tray im referring to blob tray connected to the servo. 
[image]

closed 2024-07-10 by Dendrowen