Carrot-collective/Blobifier #28: Blobifier without the Extras?

Durahl, 2024-08-12:
Is it possible to run Blobifier without the rest like the Purge Bucket getting shaken, the activation of a Servo, etc...?

If so, how could one accomplish this?

Dendrowen, 2024-08-21:
There is a SW only blobifier.cfg file in the repo that uses the bed for blobbing and knocks it off using the toolhead. The shaking can be disabled in the default version with `variable_bucket_shake_frequency: 0`

closed 2024-08-21 by Dendrowen

Durahl, 2024-08-25:
Uhm... Is the `SW only Blobifier.cfg` a Standalone thing or does it work in combination with parts of the HH Setup? 🤨

My understanding after reading through `SW only Blobifier.cfg` is that while I **_am_** able to define the **X-Axis** location of where to start Brushing and to stop via `variable_brush_start` and `variable_brush_width` there doesn't seem to be a Setting for where on the **Y-Axis** this is supposed to happen.

Same for where to do the Purge... There is again only a `variable_purge_X` but no `variable_purge_Y`.

Dendrowen, 2024-09-10:
I'd ask this question in the ercf_questions discord channel. The developer of the SW only roams there.