Carrot-collective/Blobifier #30: Feature Request: Add small lift to park macro when tray is higher than bed.

rykerg9, 2024-09-22:
I've had to manually add a G1 command to lift the toolhead so that it doesn't crash into the tray during the first few layers of a print. See below for my kind of jank fix. 2nd G1 command is the one I added.

```
[gcode_macro BLOBIFIER_PARK]

    ...

G1 X{bl.brush_start} Y{position_y} F{bl.travel_spd_xy}

G1 Z{bl.tray_top + 5} F{bl.travel_spd_z}

G1 X{bl.purge_x}

    ...
```

Dendrowen, 2024-10-09:
I've added a variable for the shaker arm height which should prevent this if set correctly. The update should come with the next Happy Hare update somewhere this week. Please feel free to reopen if it didn't work

closed 2024-10-09 by Dendrowen

rykerg9, 2024-10-30:
Sorry for the delay in response- it's been a crazy month for me.
I believe there has been a misunderstanding in regards to what exactly the issue was. I was having trouble with the nozzle not being lifted during the first few layers when swapping filaments, causing the nozzle to collide with the purge tray. For clarity, it would be ideal if a variable or macro could be integrated to lift the nozzle above the tray height when it is below said height before moving over to the tray.

Apologies for the confusion.